deepsealabs/libdc-swift #1: Receive data from the dive computer.

ArtCC, 2024-12-11:
Hi! Have you managed to receive data from the dive computer? I’ve managed to open communication with a Cressi Goa, but when I try to receive my logged dives, I get an error—no way around it. Thanks!

latishab, 2024-12-11:
Hi, I haven't look up to the problem again after a few months. I'll check it out again after finishing my college exams by next week. Hopefully it could be solved ASAP.

closed 2024-12-17 by ArtCC

latishab, 2024-12-24:
anyways, i have updated everything, the package is pretty much plug and play now. just need to parse all the remaining data.

ArtCC, 2024-12-27:
> anyways, i have updated everything, the package is pretty much plug and play now. just need to parse all the remaining data.

Thanks! Have you managed to export the dives on your dive computer?

latishab, 2024-12-27:
yes!

ArtCC, 2024-12-27:
> yes! i haven't fully finish getting all the decompression data. but overall the basic dive data is complete.

Great job! I have a Cressi Goa, and it gives me an error when trying to retrieve dive data. I'll get back to it after the holidays and review your code! Thanks!

latishab, 2024-12-28:
No problem, hopefully you can test out the Cressi Goa and make a PR to the repo.

ArtCC, 2025-01-03:
Hi!

In an empty project, I integrated the library, but I have errors. The repo URL in the wiki is incorrect; it’s missing the `github.com`. Then, the `import LibDCSwift` doesn’t find the library, and the `libdcswift-bridging-header.h` isn’t found either. Are there integration steps missing?

Thanks!

latishab, 2025-01-03:
Yes, sorry I haven't finish updating the Wiki.

Basically just put https://github.com/latishab/libdc-swift as the SPM package in your package dependencies.

Then do:
import LibDCBridge // this access Libdivecomputer related functions
import LibDCSwift // this access swift related functions like BLEManager

ArtCC, 2025-01-03:
Yes, I know, thanks. But import LibDCSwift isn't found...

latishab, 2025-01-03:
Hmm, it works on my end, have you import those two packages? Check your build settings and make sure they are there.

ArtCC, 2025-01-03:
Yes, everything seems correct...

latishab, 2025-01-04:
Hmm, turns out SPM packages don't accept bridging headers due to conflicts with client code. I have updated it, please update your package to v1.2.1. You should be able to do it now, don't import LibDCBridge to your target, just the LibDCSwift.

ArtCC, 2025-01-04:
Excellent work, thank you very much. I just integrated the library; I’ll see if I can sync with my Cressi Goa.