deepsealabs/libdc-swift #6: Add example test app

latishab, 2025-01-03:
The package currently lacks of example code that shows on how to use it. Things to emphasize on the example test app would be on how to use the LibDCSwift and LibDCBridge.

smeys, 2025-01-14:
Hi, do you have by any chance an update on the example app? I'm trying to communicate with my computer without much success and I'm new to swift. Cheers and nice work.

latishab, 2025-01-17:
hi, I do have the example app. but I haven't update it, probably this weekend.

smeys, 2025-01-26:
Hey, 
Any update on the example app?
Looking forward to give it a try.

smeys, 2025-02-01:
Hi again,

I finally managed the first step, running the `.startScanning()` is seems that the **Quick Start** guide should say:
`let manager = CoreBluetoothManager.sharedManager` instead of `let manager = CoreBluetoothManager.shared`.

Now I'm stuck in the second step, connect to the device. I'm trying to connect to my shearwater Tern using:
`Button("Connect") {
                let success = DeviceConfiguration.openBLEDevice(
                    name: "Tern",
                    deviceAddress: "fe25c237-0ece-443c-b0aa-e02033e7029d")
}`

(Computer is identified as Tern: 🔍 DEBUG [3:17:40.304 PM] [BLEManager.swift] Discovered Tern)

But I'm getting this error:
🔍 DEBUG [3:17:48.171 PM] [DeviceConfiguration.swift] Attempting to open BLE device: Tern at address: fe25c237-0ece-443c-b0aa-e02033e7029d
Failed to connect to device
ble_packet_open: Failed to connect to device
Failed to open BLE connection, rc=-6
❌ ERROR [3:17:48.186 PM] [DeviceConfiguration.swift] Failed to open device (status: dc_status_t(rawValue: -6))

Any idea what's the problem?

latishab, 2025-02-02:
it's hard to tell which part did you have a problem with. but you may want to isolate this part and tell for yourselves:
`/*--------------------------------------------------------------------
 * Opens a BLE device using a provided descriptor
 * 
 * @param data:       Pointer to device_data_t to store device info
 * @param devaddr:    BLE device address/UUID
 * @param descriptor: Device descriptor for the dive computer
 * 
 * @return: DC_STATUS_SUCCESS on success, error code otherwise
 * @note: Takes ownership of the device_data_t structure
 *------------------------------------------------------------------*/
dc_status_t open_ble_device(device_data_t *data, const char *devaddr, dc_family_t family, unsigned int model) {
    dc_status_t rc;
    dc_descriptor_t *descriptor = NULL;

    if (!data || !devaddr) {
        return DC_STATUS_INVALIDARGS;
    }

    // Initialize all pointers to NULL
    memset(data, 0, sizeof(device_data_t));
    
    // Create context
    rc = dc_context_new(&data->context);
    if (rc != DC_STATUS_SUCCESS) {
        printf("Failed to create context, rc=%d\n", rc);
        return rc;
    }

    // Get descriptor for the device
    rc = find_descriptor_by_model(&descriptor, family, model);
    if (rc != DC_STATUS_SUCCESS) {
        printf("Failed to find descriptor, rc=%d\n", rc);
        close_device_data(data);
        return rc;
    }

    // Create BLE iostream
    rc = ble_packet_open(&data->iostream, data->context, devaddr, data);
    if (rc != DC_STATUS_SUCCESS) {
        printf("Failed to open BLE connection, rc=%d\n", rc);
        close_device_data(data);
        return rc;
    }

    // Use dc_device_open to handle device-specific opening
    rc = dc_device_open(&data->device, data->context, descriptor, data->iostream);
    if (rc != DC_STATUS_SUCCESS) {
        printf("Failed to open device, rc=%d\n", rc);
        close_device_data(data);
        return rc;
    }

    // Set up event handler
    unsigned int events = DC_EVENT_DEVINFO | DC_EVENT_PROGRESS | DC_EVENT_CLOCK;
    rc = dc_device_set_events(data->device, events, event_cb, data);
    if (rc != DC_STATUS_SUCCESS) {
        printf("Failed to set event handler, rc=%d\n", rc);
        close_device_data(data);
        return rc;
    }

    // Store the descriptor
    data->descriptor = descriptor;

    // Store model string from descriptor
    if (descriptor) {
        const char *vendor = dc_descriptor_get_vendor(descriptor);
        const char *product = dc_descriptor_get_product(descriptor);
        if (vendor && product) {
            // Allocate space for "Vendor Product"
            size_t len = strlen(vendor) + strlen(product) + 2;  // +2 for space and null terminator
            char *full_name = malloc(len);
            if (full_name) {
                snprintf(full_name, len, "%s %s", vendor, product);
                data->model = full_name;  // Store full name
            }
        }
    }

    return DC_STATUS_SUCCESS;
}
`

but judging from your errors alone, I can tell you setup the SwiftUI app and the livdc-swift API wrong. I'll try to finish the example app for you to try today.