deepsealabs/libdc-swift #8: Improve device name normalization for Bluetooth device matching

latishab, 2025-01-03:
Currently, the device name normalization process in libdivecomputer has some limitations when matching Bluetooth device names to device descriptors:

1. Device descriptors use short names (e.g. "D5") while Bluetooth devices broadcast longer names with manufacturer prefix and serial numbers (e.g. "Suunto D5 12345678")

2. The `find_matching_descriptor()` function fails to match these longer names because it looks for exact matches with the product names in the descriptor list

3. The fallback string parsing in `normalizeDeviceType()` is basic and may not handle all manufacturer naming patterns correctly

Example cases:
- Suunto: "Suunto D5 12345678" should match descriptor "D5"
- Shearwater: "Petrel 2 1234" should match "Petrel 2"
- Mares: "Mares Genius 5678" should match "Genius"

Suggested improvements:

1. Add manufacturer-specific name parsing rules that understand common patterns:
   - Known manufacturer prefixes ("Suunto ", "Mares ", etc.)
   - Serial number patterns (trailing digits/hex)
   - Model variants (e.g. "2", "Air", "Pro")

2. Update the descriptor matching to:
   - Try exact matches first
   - Then try matching without manufacturer prefix
   - Then try matching the normalized name
   - Consider fuzzy matching for close matches

3. Add test cases for various real-world device name formats

This would improve the reliability of device identification, especially for Bluetooth devices where we have less control over the advertised names.

Related files:
- src/descriptor.c
- include/descriptor.h

ArtCC, 2025-01-04:
Indeed, an example is with the Cressi Goa. In my case, its BLE device name is 2_466a. If you directly search for the descriptor, it returns an incorrect computer. In my case, in previous tests, I solved it by first selecting the brand and model of the computer, then using that data to scan with BLE and filter.

In the case of your library, the `openBLEDevice` function returns an error because it detects an incorrect computer.

I am not a C developer, and I find it difficult to understand the internal workings of the `libdivecomputer` library, but I would like to help in any way I can to make it work with Cressi dive computers.

latishab, 2025-01-04:
Could you try to send me the logs shown on your console? I'll see what I can do.

ArtCC, 2025-01-04:
> Could you try to send me the logs shown on your console? I'll see what I can do too. Thanks for the help! It's quite hard to verify for all computer brands because I only have Suunto.

Of course! At the moment, I’m not receiving many logs. I connect to my computer correctly, but as I mentioned, it doesn’t identify it properly.

```
ℹ️ INFO [17:28:04.841] [BLEManager.swift] Bluetooth is powered on
🔍 DEBUG [17:28:19.976] [DeviceConfiguration.swift] Attempting to open BLE device: 2_466a at address: 67A48076-99C4-8...
🔍 DEBUG [17:28:19.977] [StoredDevice.swift] No stored devices data found in UserDefaults
❌ ERROR [17:28:19.998] [DeviceConfiguration.swift] Failed to identify device: dc_status_t(rawValue: -1)
ℹ️ INFO [17:28:19.999] [BLEManager.swift] Successfully connected to 2_466a
```

In my case, I solved it as I mentioned, by filtering by brand, model, and BLE name.

```
- (NSDictionary<NSString *, NSString *> *)getDiveComputerForBrand:(NSString *)brand
                                                            model:(NSString *)model
                                                       deviceName:(NSString *)deviceName {
    dc_iterator_t *iterator = NULL;
    dc_status_t status = dc_descriptor_iterator(&iterator);

    if (status != DC_STATUS_SUCCESS) {
        NSLog(@"Error initializing the descriptor iterator.");

        return nil;
    }

    dc_descriptor_t *descriptor = NULL;

    while (dc_iterator_next(iterator, (void **)&descriptor) == DC_STATUS_SUCCESS) {
        const char *vendor = dc_descriptor_get_vendor(descriptor);
        const char *product = dc_descriptor_get_product(descriptor);

        if (vendor && product && [brand isEqualToString:@(vendor)] && [model isEqualToString:@(product)]) {
            if (dc_descriptor_filter(descriptor, DC_TRANSPORT_BLE, [deviceName UTF8String])) {
                unsigned int modelNumber = dc_descriptor_get_model(descriptor);

                dc_iterator_free(iterator);

                return @{
                    @"brand": @(vendor),
                    @"model": @(product),
                    @"modelNumber": [NSString stringWithFormat:@"%u", modelNumber]
                };
            }
        }
    }

    dc_iterator_free(iterator);

    return nil;
}
```

latishab, 2025-01-04:
Thanks for the info! I have made a lot of changes on configuredc.c. Now it matches name using dc_descriptor_filter(). Could you try bluetooth-device-matching branch with your Cressi device?

Let me know if it works.

ArtCC, 2025-01-04:
> Thanks for the info! I have made a lot of changes on configuredc.c. Now it matches name using dc_descriptor_filter(). Could you try bluetooth-device-matching branch with your Cressi device?
> 
> Let me know if it works.

Of course!

In the function:

```
dc_status_t find_descriptor_by_name(dc_descriptor_t **out_descriptor, const char *name) {}
```

It correctly detects my Cressi Goa, but in the line:

```
if (product && strstr(name, product) != NULL) {}
```

It doesn't enter because, in Cressi devices, the name (2_466a) is not part of the product (Goa), so it returns:

```
return DC_STATUS_UNSUPPORTED;
```

I believe Jef Driesen mentioned this case to me, or I might be remembering it wrong. I think in this case, when vendor, product, and family are correct, the model is 2_ (model 2 = Goa).

latishab, 2025-01-05:
Hmm, yeah I noticed that. I noticed how descriptor.c have these dc_match_name() and dc_match_prefix(). But I couldn't figure out how it couldn't find the exact match.

So I ended up surfing through subsurface's [bluetooth discovery algorithm](https://github.com/subsurface/subsurface/blob/972b7a0643f0b5c98253c5dc4ad43951668f04a2/core/btdiscovery.cpp#L111-L163), and came up with the solution that there could be 3 ways to do it; match exact name, match substring, or match prefix just like your use case.

Can you try out the bluetooth-device-matching new update? Let me know if it works, and if something doesn't work in your use case, please update the find_descriptor_by_name algorithm a bit.

ArtCC, 2025-01-05:
From what I see, it now correctly detects the Cressi Goa family and continues the process, reaching:

```
ℹ️ INFO [11:02:10.419] [BLEManager.swift] Successfully connected to 2_466a  
🔍 DEBUG [11:02:12.481] [DeviceConfiguration.swift] Attempting to open BLE device: 2_466a at address: 67A48076-99C4-81F9-F...  
🔍 DEBUG [11:02:12.482] [StoredDevice.swift] No stored devices data found in UserDefaults  
```

In the function:

```
dc_status_t open_ble_device(device_data_t *data, const char *devaddr, dc_family_t family, unsigned int model) {}

rc = find_descriptor_by_model(&descriptor, family, model);
```

It's correct; I receive:

```
(lldb) print family  
(dc_family_t) DC_FAMILY_CRESSI_GOA  
```

But in:

```
if (!connectToBLEDevice(io, devaddr)) {}
```

It seems to get stuck at:

```
success = [manager discoverServices];
```

I'm looking to see if I can find anything else.

ArtCC, 2025-01-05:
From what I see, it gets stuck in the loop and doesn’t search for the services.

```
@objc(discoverServices)
    public func discoverServices() -> Bool {
        guard let peripheral = self.peripheral else { return false }
        peripheral.discoverServices(nil)
        while writeCharacteristic == nil || notifyCharacteristic == nil {
            RunLoop.current.run(until: Date(timeIntervalSinceNow: 0.05))
        }
        
        return writeCharacteristic != nil && notifyCharacteristic != nil
    }
```

latishab, 2025-01-05:
Hmm, the problem is, it works well in my side, could the problem is the BLEManager didn't detect your Cressi's services? Because I have put all the possible services in DeviceConfiguration based on Subsurface's [qt-ble.cpp](https://github.com/subsurface/subsurface/blob/39e47845d7c494571c87a09e4a0216ea6ed845c0/core/qt-ble.cpp).

ArtCC, 2025-01-05:
A question, do we need to do something before?:

```
DeviceConfiguration.openBLEDevice(name: deviceName, deviceAddress: bleAddress)
```

In this case, I receive that it's not fully connected. I notice a 1-second delay, but it might not be enough to proceed with sending messages:

```
API MISUSE: <CBPeripheral: 0x303dc8340, identifier = 67A48076-99C4-81F9-F3B0-74C121A15E6D, name = 2_466a, mtu = 0, state = connecting> can only accept commands while in the connected state
```

ArtCC, 2025-01-05:
No, that's not it.

latishab, 2025-01-05:
`//
//  DeviceRow.swift
//  BLETest
//
//  Created by User on 24/12/2024.
//

import Foundation
import SwiftUI
import CoreBluetooth
import LibDCSwift

struct DeviceRow: View {
    let device: CBPeripheral?
    let storedDevice: StoredDevice?
    @ObservedObject var bluetoothManager: CoreBluetoothManager
    @ObservedObject var diveViewModel: DiveDataViewModel
    @Binding var showConnectedDeviceSheet: Bool
    @State private var isConnecting = false
    
    // Cache these values to avoid recomputing during rendering
    private let displayName: String
    private let deviceIdentity: (family: DeviceConfiguration.DeviceFamily, model: UInt32)?
    
    init(device: CBPeripheral?,
         storedDevice: StoredDevice?,
         bluetoothManager: CoreBluetoothManager,
         diveViewModel: DiveDataViewModel,
         showConnectedDeviceSheet: Binding<Bool>) {
        self.device = device
        self.storedDevice = storedDevice
        self.bluetoothManager = bluetoothManager
        self.diveViewModel = diveViewModel
        self._showConnectedDeviceSheet = showConnectedDeviceSheet
        
        // Compute these once during initialization
        if let device = device {
            self.displayName = device.name ?? "Unknown Device"
            if let name = device.name {
                self.deviceIdentity = DeviceConfiguration.fromName(name)
            } else {
                self.deviceIdentity = nil
            }
        } else {
            self.displayName = storedDevice?.name ?? "Unknown Device"
            self.deviceIdentity = nil
        }
    }
    
    private var isDeviceSupported: Bool {
        deviceIdentity != nil
    }
    
    private var isConnected: Bool {
        bluetoothManager.connectedDevice?.identifier == device?.identifier &&
        bluetoothManager.isPeripheralReady
    }
    
    private var buttonState: ButtonState {
        if isConnecting {
            return .connecting
        } else if isConnected {
            return .viewDetails
        } else if bluetoothManager.connectedDevice?.identifier == device?.identifier {
            return .connecting
        } else if bluetoothManager.isDisconnecting && bluetoothManager.connectedDevice?.identifier == device?.identifier {
            return .disconnecting
        } else {
            return storedDevice != nil ? .sync : .connect
        }
[... 132 lines cut ...]
            connectionTimer?.invalidate()
            connectionTimer = nil
            if isConnecting {
                isConnecting = false
            }
        }
        .onReceive(bluetoothManager.$isPeripheralReady) { ready in
            if ready && bluetoothManager.connectedDevice?.identifier == device?.identifier {
                isConnecting = false
                connectionTimer?.invalidate()
                connectionTimer = nil
            }
        }
        .onReceive(bluetoothManager.$connectedDevice) { connectedDevice in
            if connectedDevice?.identifier != device?.identifier && isConnecting {
                isConnecting = false
                connectionTimer?.invalidate()
                connectionTimer = nil
            }
        }
    }
}
`

This is what I do to connect. Note that you should not use other functions in BLEManager for connecting other than the DeviceConfiguration.openBLEDevice()

ArtCC, 2025-01-05:
Honestly, I'm not doing anything different. Well, I'll keep reviewing to see if I can figure it out. Thanks!

ArtCC, 2025-01-05:
Got it now, but it seems that receiving the data is very slow, and after a while, the computer disconnects the Bluetooth interface. Does this happen to you?

```
ℹ️ INFO [13:47:31.832] [BLEManager.swift] Successfully connected to 2_466a
ℹ️ INFO [13:47:32.840] [BLEManager.swift] Found known service: Nordic Semi UART
ℹ️ INFO [13:47:32.841] [BLEManager.swift] Discovering characteristics for unknown service: 6E400001-B5A3-F393-E0A9-E50E24DC10B8
ℹ️ INFO [13:47:32.842] [BLEManager.swift] Found notify characteristic: Battery Level
ℹ️ INFO [13:47:32.843] [BLEManager.swift] Found write characteristic: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E
ℹ️ INFO [13:47:32.843] [BLEManager.swift] Found notify characteristic: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E
ℹ️ INFO [13:47:32.843] [BLEManager.swift] Found write characteristic: 6E400001-B5A3-F393-E0A9-E50E24DC10B8
ℹ️ INFO [13:47:32.844] [BLEManager.swift] Found notify characteristic: 6E400002-B5A3-F393-E0A9-E50E24DC10B8
ℹ️ INFO [13:47:32.844] [BLEManager.swift] Found write characteristic: 8EC90003-F315-4F60-9FB8-838830DAEA50
ℹ️ INFO [13:47:32.844] [BLEManager.swift] Found notify characteristic: 8EC90003-F315-4F60-9FB8-838830DAEA50
ℹ️ INFO [13:47:32.888] [BLEManager.swift] Notification state updated: enabled
ℹ️ INFO [13:47:32.947] [BLEManager.swift] Notification state updated: enabled
🔍 DEBUG [13:47:32.993] [DeviceConfiguration.swift] Successfully opened device
🔍 DEBUG [13:47:33.001] [DeviceConfiguration.swift] Device data pointer allocated at: 0x0000000302957300
ℹ️ Info: ExternalDiveSessionViewModel: openDevice: Cressi, Goa, 2_466a
🔍 DEBUG [13:47:33.024] [BLEManager.swift] Device data pointer set
ℹ️ INFO [13:47:33.024] [BLEManager.swift] Notification state updated: enabled
ℹ️ INFO [13:47:33.068] [BLEManager.swift] Notification state updated: enabled
ℹ️ INFO [13:47:33.127] [BLEManager.swift] Notification state updated: enabled
🔍 DEBUG [13:50:00.606] [BLEManager.swift] Received data: 5c... (1 bytes)
🔍 DEBUG [13:50:00.607] [BLEManager.swift] Buffer: 5c
```

ArtCC, 2025-01-05:
Ok.

```
ℹ️ INFO [14:17:21.534] [DiveLogRetriever.swift] 💡 No stored fingerprint - downloading all dives
ℹ️ INFO [14:17:21.534] [DiveLogRetriever.swift] 🔄 Starting dive enumeration...
❌ ERROR [14:17:21.542] [DiveLogRetriever.swift] ❌ Download failed or was interrupted
💣 Error: DiveComputerManager: retrieveDiveLogs: failure
```

latishab, 2025-01-05:
this is how i retrieve my dive logs once the device pointer is set btw:
`private func retrieveDiveLogs() {
        logInfo("🎯 Starting dive log retrieval for device: \(device.name ?? "Unknown")")
        guard let devicePtr = bluetoothManager.openedDeviceDataPtr else {
            logError("❌ Device data pointer is nil during retrieveDiveLogs")
            errorMessage = "No device connection found"
            showError = true
            return
        }
        
        bluetoothManager.isRetrievingLogs = true
        bluetoothManager.currentRetrievalDevice = device
        
        // Request background execution time
        #if os(iOS)
        var backgroundTask: UIBackgroundTaskIdentifier = .invalid
        backgroundTask = UIApplication.shared.beginBackgroundTask {
            if backgroundTask != .invalid {
                UIApplication.shared.endBackgroundTask(backgroundTask)
                backgroundTask = .invalid
            }
        }
        #endif
        
        // Keep the Bluetooth connection alive in background
        bluetoothManager.setBackgroundMode(true)
        
        DiveLogRetriever.retrieveDiveLogs(
            from: devicePtr,
            device: device,
            viewModel: diveViewModel,
            bluetoothManager: bluetoothManager,
            onProgress: { current, total in
                DispatchQueue.main.async {
                    diveViewModel.updateProgress(current: current + 1)
                    buttonRefreshID = UUID()
                }
            },
            completion: { success in
                DispatchQueue.main.async {
                    if case .noNewDives = diveViewModel.progress {
                        let lastUpdate = diveViewModel.getFingerprintInfo(
                            forDeviceType: deviceDetails.modelName,
                            serial: deviceDetails.serial
                        )?.formatted(date: .long, time: .shortened) ?? "previous download"
                        
                        noDivesMessage = if diveViewModel.getFingerprint(
                            forDeviceType: deviceDetails.modelName,
                            serial: deviceDetails.serial
                        ) != nil {
                            """
                            No new dives found since \(lastUpdate).
                            
                            To download all dives again, disable "Download Only New Dives" in Settings.
                            """
                        } else {
                            "No dives found on device."
                        }
                        showNoDivesAlert = true
                    } else if !success {
                        self.errorMessage = self.diveViewModel.progress.description
                        self.showError = true
                    }
                    
                    if bluetoothManager.currentRetrievalDevice?.identifier == device.identifier {
                        bluetoothManager.clearRetrievalState()
                    }
                    
                    // Reset background mode
                    bluetoothManager.setBackgroundMode(false)
                    buttonRefreshID = UUID()
                    
                    #if os(iOS)
                    if backgroundTask != .invalid {
                        UIApplication.shared.endBackgroundTask(backgroundTask)
                        backgroundTask = .invalid
                    }
                    #endif
                }
            }
        )`

also, here are the things you should have expect when retrieving dive logs:
ℹ️ INFO [21:22:46.499] [ConnectedDeviceView.swift] 🎯 Starting dive log retrieval for device: Suunto D5 204710004359
ℹ️ INFO [21:22:46.502] [BLEManager.swift] Setting up background mode for peripheral: 88DCDDF8-9FF0-3587-F887-B4597215F1E0
ℹ️ INFO [21:22:46.503] [DiveLogRetriever.swift] 💡 No stored fingerprint - downloading all dives
ℹ️ INFO [21:22:46.503] [DiveLogRetriever.swift] 🔄 Starting dive enumeration...
ℹ️ INFO [21:22:46.503] [DiveLogRetriever.swift] ❌ No stored fingerprint found for Suunto D5 (a9aac687)
No fingerprint found
ℹ️ INFO [21:23:00.416] [DiveLogRetriever.swift] 📍 New fingerprint from latest dive: 0ee51b66
ℹ️ INFO [21:23:00.426] [DiveLogRetriever.swift] ✅ Parsed dive #1
ℹ️ INFO [21:23:12.634] [DiveLogRetriever.swift] ✅ Parsed dive #2

ArtCC, 2025-01-05:
In line:

```
let enumStatus = dc_device_foreach(dcDevice, diveCallbackClosure, contextPtr)
```

Received:

```
enumStatus	dc_status_t	.DC_STATUS_UNSUPPORTED	

Printing description of enumStatus:
▿ dc_status_t
  - rawValue : -1
```

ArtCC, 2025-01-05:
From what I see, in the function:

```
static dc_status_t
cressi_goa_device_foreach (dc_device_t *abstract, dc_dive_callback_t callback, void *userdata)
```

It enters the error here:

```
status = dc_iostream_ioctl (device->iostream, DC_IOCTL_BLE_CHARACTERISTIC_READ, request, sizeof(dc_ble_uuid_t) + sizes[i]);
			if (status != DC_STATUS_SUCCESS) {
				char uuidstr[DC_BLE_UUID_SIZE] = {0};
				ERROR (abstract->context, "Failed to read the characteristic '%s'.",
					dc_ble_uuid2str(characteristics[i], uuidstr, sizeof(uuidstr)));
				goto error_free_id;
			}
```

I don’t know the reason because, according to the `libdivecomputer` library, this service is correct:

```
ℹ️ INFO [19:35:02.409] [BLEManager.swift] Found known service: Nordic Semi UART
ℹ️ INFO [19:35:02.410] [BLEManager.swift] Discovering characteristics for unknown service: 6E400001-B5A3-F393-E0A9-E50E24DC10B8
```

Do you have any ideas? At this point, I had issues with my code before seeing your library, although I probably had some other problems during the connection as well. I'm starting to think Cressi is problematic...

latishab, 2025-01-06:
I'm not sure if Cressi is problematic. Have you had issues just from the iOS Bluetooth settings itself? Do you get disconnected all of the sudden? If not then Cressi isn't problematic, If yes; then probably Cressi is problematic.

I saw your logs previously here:
ℹ️ INFO [13:47:31.832] [BLEManager.swift] Successfully connected to 2_466a
ℹ️ INFO [13:47:32.840] [BLEManager.swift] Found known service: Nordic Semi UART
ℹ️ INFO [13:47:32.841] [BLEManager.swift] Discovering characteristics for unknown service: 6E400001-B5A3-F393-E0A9-E50E24DC10B8
ℹ️ INFO [13:47:32.842] [BLEManager.swift] Found notify characteristic: Battery Level
ℹ️ INFO [13:47:32.843] [BLEManager.swift] Found write characteristic: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E
ℹ️ INFO [13:47:32.843] [BLEManager.swift] Found notify characteristic: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E
ℹ️ INFO [13:47:32.843] [BLEManager.swift] Found write characteristic: 6E400001-B5A3-F393-E0A9-E50E24DC10B8
ℹ️ INFO [13:47:32.844] [BLEManager.swift] Found notify characteristic: 6E400002-B5A3-F393-E0A9-E50E24DC10B8
ℹ️ INFO [13:47:32.844] [BLEManager.swift] Found write characteristic: 8EC90003-F315-4F60-9FB8-838830DAEA50
ℹ️ INFO [13:47:32.844] [BLEManager.swift] Found notify characteristic: 8EC90003-F315-4F60-9FB8-838830DAEA50
ℹ️ INFO [13:47:32.888] [BLEManager.swift] Notification state updated: enabled
ℹ️ INFO [13:47:32.947] [BLEManager.swift] Notification state updated: enabled
🔍 DEBUG [13:47:32.993] [DeviceConfiguration.swift] Successfully opened device
🔍 DEBUG [13:47:33.001] [DeviceConfiguration.swift] Device data pointer allocated at: 0x0000000302957300

I believe once you notification state are enabled, the device is opened, and device data pointer is allocated, you are good to go. You may want to download LightBlue app to verify your services and characteristics UUID. For a BLE device, there should be a read characteristic, and a write but no response characteristic.

latishab, 2025-01-06:
You also may want to take a look at these functions for better tracing:

open_ble_device -> dc_device_open -> cressi_goa_device_open -> dc_device_foreach

latishab, 2025-01-10:
Have you solved it? Anything that I can help?

ArtCC, 2025-01-10:
> Have you solved it? Anything that I can help?

Honestly, I haven't looked at it again. I'll dedicate some time to it this weekend. I can only test with my Cressi Goa, and I’m not very skilled with C code. If the issue is more related to Cressi than to the code, I’m not sure what else I could do.  

My project is intended for use with more dive computers, so I’m trying to find someone who could test it with theirs if it’s not Cressi. Although, of course, I want to use my project.

latishab, 2025-01-23:
Give me your testflight and i'll test yours with my Suunto D5 (i'm about to sell it for a peregrine, better now or never). Also i would also need your help to test mine as well.

ArtCC, 2025-01-27:
> Have you solved it? Anything that I can help?

I've been checking, and with LightBlue, I see it has these services:

0x6Е400002-B5A3-F393-E0A9-E50E24DCCA9E
Properties: Write, Write without Response

0x6Е400003-B5A3-F393-E0A9-E50E24DCCA9E
Properties: Notify, Read

And:

0x6Е400001-B5A3-F393-E0A9-E50E24DC10B8
Properties: Write, Write without Response

0x6E400002-B5A3-F393-E0A9-E50E24DC10B8
Properties: Notify

0x6Е400003-B5A3-F393-E0A9-E50E24DC10B8
Properties: Read

0x6E400004-B5A3-F393-E0A9-E50E24DC10B8
Properties: Read

0x6Е400005-B5A3-F393-E0A9-E50E24DC10B8
Properties: Read

Your library uses the first ones, but then in the cressi_goa_device_foreach code I see:

const dc_ble_uuid_t characteristics[] = {
	{0x6E, 0x40, 0x00, 0x03, 0xB5, 0xA3, 0xF3, 0x93, 0xE0, 0xA9, 0xE5, 0x0E, 0x24, 0xDC, 0x10, 0xB8},
	{0x6E, 0x40, 0x00, 0x04, 0xB5, 0xA3, 0xF3, 0x93, 0xE0, 0xA9, 0xE5, 0x0E, 0x24, 0xDC, 0x10, 0xB8},
	{0x6E, 0x40, 0x00, 0x05, 0xB5, 0xA3, 0xF3, 0x93, 0xE0, 0xA9, 0xE5, 0x0E, 0x24, 0xDC, 0x10, 0xB8},
};

And of course, I understand that when iterating over that array, they don't match, and it throws an error—or so I think. But I've made changes and tests, and I haven't achieved anything; it still hits the error at:

if (status != DC_STATUS_SUCCESS) {
	char uuidstr[DC_BLE_UUID_SIZE] = {0};
	ERROR (abstract->context, "Failed to read the characteristic '%s'.",
		dc_ble_uuid2str(characteristics[i], uuidstr, sizeof(uuidstr)));
	goto error_free_id;
}

Do you think of anything? Did you have these issues with your Suunto computer?

latishab, 2025-01-28:
Seems like for that array of dc_ble_uuid_t_characteristics[], it uses characteristics ending in 10B8. Your Cressi Goa has services with UUIDs ending in CA9E and 10B8, perhaps you can try to subscribe to the characteristics ending with 10B8 and see if it works.

ArtCC, 2025-01-28:
> Seems like for that array of dc_ble_uuid_t_characteristics[], it uses characteristics ending in 10B8. Your Cressi Goa has services with UUIDs ending in CA9E and 10B8, perhaps you can try to subscribe to the characteristics ending with 10B8 and see if it works.
> 
> I have no problem with my Suunto D5 because it only has 2 services; 1 service for notify, 1 service for write without response. Given that you are having problem with cressi_goa_device_foreach, does the device open function in DeviceConfiguration works for you?

Yes, I tested it yesterday, but it still had the same issue. I'll keep checking.

latishab, 2025-01-28:
So are you able to use func "openBLEDevice(name: String, deviceAddress: String)" or no? So I can try to narrow down where the problem at.

ArtCC, 2025-01-29:
> So are you able to use func "openBLEDevice(name: String, deviceAddress: String)" or no? So I can try to narrow down where the problem at.

Yes. There don't seem to be any issues with that. According to the libdivecomputer library, the problem arises when using Bluetooth features, but I’ve never gotten past that point.

🔍 DEBUG [13:47:32.993] [DeviceConfiguration.swift] Successfully opened device
🔍 DEBUG [13:47:33.001] [DeviceConfiguration.swift] Device data pointer allocated at: 0x0000000302957300
ℹ️ Info: ExternalDiveSessionViewModel: openDevice: Cressi, Goa, 2_466a
🔍 DEBUG [13:47:33.024] [BLEManager.swift] Device data pointer set

jefdriesen, 2025-03-11:
For Cressi, you need to ignore the Nordic UART Service and use the Cressi service. See the documentation here:
https://github.com/libdivecomputer/libdivecomputer/wiki/BLE

You'll also need to implement the `DC_IOCTL_BLE_CHARACTERISTIC_READ` ioctl because libdivecomputer uses that api to read the other 3 Cressi characteristics to obtain the serial/model, firmware, etc.

ArtCC, 2025-03-11:
> For Cressi, you need to ignore the Nordic UART Service and use the Cressi service. See the documentation here: https://github.com/libdivecomputer/libdivecomputer/wiki/BLE
>
> You'll also need to [...]

Hi Jef, nice to greet you again. The first part in Latishab's code, as I understand it, is to add the service in:

```
@objc private let knownSerialServices: [SerialService] = [
        SerialService(uuid: "0000fefb-0000-1000-8000-00805f9b34fb", vendor: "Heinrichs-Weikamp", product: "Telit/Stollmann"),
        SerialService(uuid: "2456e1b9-26e2-8f83-e744-f34f01e9d701", vendor: "Heinrichs-Weikamp", product: "U-Blox"),
        SerialService(uuid: "544e326b-5b72-c6b0-1c46-41c1bc448118", vendor: "Mares", product: "BlueLink Pro"),
        SerialService(uuid: "6e400001-b5a3-f393-e0a9-e50e24dcca9e", vendor: "Nordic Semi", product: "UART"),
        SerialService(uuid: "98ae7120-e62e-11e3-badd-0002a5d5c51b", vendor: "Suunto", product: "EON Steel/Core"),
        SerialService(uuid: "cb3c4555-d670-4670-bc20-b61dbc851e9a", vendor: "Pelagic", product: "i770R/i200C"),
        SerialService(uuid: "ca7b0001-f785-4c38-b599-c7c5fbadb034", vendor: "Pelagic", product: "i330R/DSX"),
        SerialService(uuid: "fdcdeaaa-295d-470e-bf15-04217b7aa0a0", vendor: "ScubaPro", product: "G2/G3"),
        SerialService(uuid: "fe25c237-0ece-443c-b0aa-e02033e7029d", vendor: "Shearwater", product: "Perdix/Teric"),
        SerialService(uuid: "0000fcef-0000-1000-8000-00805f9b34fb", vendor: "Divesoft", product: "Freedom")
    ]
```

Since right now it doesn't detect the correct characteristics because for Cressi they are not the same as for Nordic, but the second part, being C code, is a bit beyond me. Would `DC_IOCTL_BLE_CHARACTERISTIC_READ` need to be added when initially connecting to the Cressi computer?

ArtCC, 2025-03-11:
Nothing, I've reviewed it, but the C language is completely beyond my knowledge. What needs to be implemented is in:

dc_status_t ble_ioctl(ble_object_t *io, unsigned int request, void *data, size_t size) {
    return DC_STATUS_UNSUPPORTED;
}

What 2opremio has implemented in: https://github.com/subsurface/subsurface/pull/4463/files

I think this would be the correct approach for it to work with Cressi. On the other hand, I see that when using the discovered write and notification services in CoreBluetoothManager, it is not correct because it should be filtered for the correct one according to the brand.

ArtCC, 2025-03-11:
I have implemented the **ble_ioctl** function with some additional changes, but it seems that it is not enough.

```
dc_status_t ble_ioctl(ble_object_t *io, unsigned int request, void *data, size_t size) {
    if (!io) {
        return DC_STATUS_INVALIDARGS;
    }

    Class CoreBluetoothManagerClass = NSClassFromString(@"CoreBluetoothManager");
    id<CoreBluetoothManagerProtocol> manager = [CoreBluetoothManagerClass shared];

    switch (request) {
        case DC_IOCTL_BLE_GET_NAME:
            return [manager getPeripheralName:(char *)data maxLength:size] ? DC_STATUS_SUCCESS : DC_STATUS_UNSUPPORTED;
        case DC_IOCTL_BLE_CHARACTERISTIC_READ: {
            if (size < sizeof(quint128)) {
                return DC_STATUS_INVALIDARGS;
            }

            quint128 uuid;
            memcpy(uuid.data, data, sizeof(uuid.data));

            NSData *uuidData = [NSData dataWithBytes:uuid.data length:sizeof(uuid.data)];
            CBUUID *characteristicUUID = [CBUUID UUIDWithData:uuidData];

            size_t readsize = size - sizeof(uuid.data);
            char *p = ((char *)data) + sizeof(uuid.data);

            BOOL success = [manager readCharacteristic:characteristicUUID intoBuffer:p maxLength:readsize];

            return success ? DC_STATUS_SUCCESS : DC_STATUS_UNSUPPORTED;
        }
        default:
            return DC_STATUS_UNSUPPORTED;
    }
}
```

In my Cressi Goa, it uses the 3 correct characteristics according to the documentation, but then in **dc_device_foreach**, it returns **DC_STATUS_IO**, which doesn’t make much sense to me.

Any idea, @jefdriesen?

ArtCC, 2025-03-12:
@latishab Although I didn’t end up using your library, I’ve successfully imported data from my Cressi dive computer. When I have time again, I’ll review your code to add what’s missing and make some additional changes.

latishab, 2025-03-12:
@ArtCC sure thing, thanks for your help. please put a PR whenever you like and i'll review it.

jefdriesen, 2025-03-14:
> In my Cressi Goa, it uses the 3 correct characteristics according to the documentation, but then in **dc_device_foreach**, it returns **DC_STATUS_IO**, which doesn’t make much sense to me.
> 
> Any idea, [@jefdriesen](https://github.com/jefdriesen)?

Do you have a log? It's difficult to tell anything useful without more detailed information. The `DC_STATUS_IO` error is not returned by the `dc_device_foreach()` function itself, but somewhere deeper in the stack.

ArtCC, 2025-03-14:
> Do you have a log? It's difficult to tell anything useful without more detailed information. The `DC_STATUS_IO` error is not returned by the `dc_device_foreach()` function itself, but somewhere [...]

No problem, it was something else, and I’ve already fixed it. I can now import correctly from my Cressi dive computer to my app. Thanks!

latishab, 2026-11-05:
Closing as superseded. BLE names now resolve against libdivecomputer's own descriptor table (#97), and the Bluetooth matching has been reworked since this was filed. If a specific computer still fails to match, please open a new issue with its advertised name.

closed 2026-11-05 by latishab